VISiON-IITK/tracker #6: Lane and pothole detection

lakshayg, 2017-01-07:


shubhgup, 2017-01-08:
HSV threshold extraction followed by morphological opening and erosion. Perspective correction and Hough line formation on resultant image. Discrimination of right and left lane lines by lane midpoint position with respect to screen center. Labelling and grouping of lines on same lane to form single bestfit lane lines. removal of non traversable region based on lanes

shubhgup, 2017-01-08:
[image]
[image]

AnayMehrotra, 2017-01-09:
[image]
[image]
[image]

Detecting lanes using splins. 
Techniques 

1. Obstacle Removal : Done using HSV thresholding.
2. Region of Interest : A custom (variable) kernel is convoluted on the image. Regions of interest have high values. Kernel used contains ones as wide as the lane and an equal number of negative ones distributed on either side, i.e. for lane of width four kernel is [-1 -1 1 1 1 1 -1 -1]
3. Spline Making : 

- 25 interest points are chosen from either side of the image. These are used to extend splines to a region at a (specified) distance from them. 
- Splines are rated on basis of their average distance from points of interest, which are then extended further.
- Longer splines are promoted, ensuring a lane is not missed.

4. Averaging : The average of 10 frames is taken at a time. This reduces fluctuations and propagates information through frames.
5. Top view : Using warp perspective transform.

Shortcomings :

- Averaging ignores bot's motion and confidence (probability) in splines. This results in a laggy responce.

pritesh1996, 2017-01-18:
Detecting Grids where lane is present
Techniques
1. Divided into Grids 20x36.
2. Mean and Std. Deviation of pixel value of 2xBLUE-GREEN of each grid.
3. Applied Threshold of Mean + Std. Deviation. The one with higher is more probable to be part of lane (and least probable to be a grass ONLY).

closed 2018-02-18 by aalapshah12297