VISiON-IITK/tracker #9: Motion Planning Within Lane Using RRT

aalapshah12297, 2017-01-07:


aalapshah12297, 2017-01-10:
[image]
Motion plainning for non-holonomic robot using RRT
Blue nodes represent the full RRT generated
Red nodes represent the chosen path to goal (co-ordinates supplied)

aalapshah12297, 2017-01-12:
Will implement incremental RRT-based non-holonomic planning within dynamic local map with Lidar-like obstacle map. Will then try to use space-partitioning or some other geometric method to reduce time complexity of nearest-neighbour search while generating RRT.

aalapshah12297, 2017-01-14:
13/01/17
[output (with tree shown).zip](https://github.com/igvc-iitk/tracker/files/706153/output.with.tree.shown.zip)
Implemented incremental RRT-based non-holonomic planning within dynamic local map with Lidar-like obstacle map. Tuned RRT parameters (node generation limit, max edge length, etc.) till fast and reliable results were generated.

aalapshah12297, 2017-01-15:
15/01/17
[image]
[image]
Debugged previous code. Generated some paths on a simple square 100x100 map with a 50x50 obstacle in the center. Plotted the paths using Matlab (figures above).

aalapshah12297, 2017-01-17:
[image]
Completed porting of previous MATLAB script to C++. Generated some paths on the same map as before and wrote a MATLAB script to plot the output. Currently creating a ROS node using the C++ code.

aalapshah12297, 2017-01-18:
Implemented potential field based correction factor to reduce jaggedness of path generated by RRT (only slightly)
Some Examples: (Blue = original path, Red = corrected path)
1. 
[image]
[image]

2.
[image]
[image]

bhuvig, 2017-01-18:
ROS node of RRT up and running. Currently works on a hard coded map while subscribing to start and end coordinates. Publishes path on a topic.

closed 2018-02-18 by aalapshah12297