VISiON-IITK/tracker #14: Waypoint Navigation using GPS

AshishBhatti, 2017-01-12:
We were able to extract coordinates from GPS and using that I attempted a test run today. We are using the GPS inbuilt magnetometer as compass. But the magnetometer data is very inaccurate. We are planning to augment our system with magnetometer from other sensors like IMU.

Hardik1496, 2017-01-14:
Tried plotting data extracted from GPS on local map using rviz. Currently not able to plot correct data facing some issues. Searching the issues to plot the data on map

deepakgangwar, 2017-01-14:
Way point navigation using encoders is completed for firebird.
Bot can go to the next waypoint with a certain minimum velocity. The final code is present in the Control repo.

closed 2018-02-18 by aalapshah12297