VISiON-IITK/tracker #15: Grass Classification

shubhgup, 2017-01-14:


AnayMehrotra, 2017-01-14:
Created a code to generate samples for training SVM.
The samples are 50px.50px, made by moving a window over manually cropped image, 10px at a time.  Horizontally and vertically flipped images are also saved.

Dataset of 20,000 positive samples of grass and 6,000 negative samples was made.

AnayMehrotra, 2017-01-14:
SVM was trained on HOG features, for grass classification.
Python script trains SVM on a dataset and stores the classifier in a file,  for future use.
Results : Unsatisfactory. SVM is unable to identify grass as of now.

shubhgup, 2017-01-14:
Grass classification using HSV values . Learning on 20000 50px X 50px images of grass to find global mean and std. deviation for each channel. These values were used to create 2*std_dev bounds around global means for which windows extracted from video frame were checked to belong to be classified as grass. These grass regions were then used to update and adapt the mean and std. deviation values for grass detection in next frame.

Currently, the algorithm is capable of detecting grass matching the learning dataset, and is also capable of learning to work in case of lighting changes. However, due to this update, it is failing to work properly with potholes as it learns pothole regions as very lighted up grass regions after a while.

[image]
[image]
[image]

closed 2018-02-18 by aalapshah12297